mxtommy/Kip #1: HTML5 browser features

netAction, 2017-10-31:
- Keep screen bright
- Fullscreen
- No history API or fix support in Signal K Node server

mxtommy, 2017-10-31:
hmmm, I had been going fullscreen via my browser's fullscreen menu option. Hadn't thought about how easy it would be to just ask the app to go fullscreen. Great idea :)

As far as keeping the screen bright/on I think that's OS dependent? Ie you need to set it to not put the monitor to sleep after X minutes. Not sure a webapp has permissions to change that, though maybe being fullscreen will prevent it?

Regarding history API, There's been some discussion on signalk's slack regarding it. Nothing's been formalized though. Waiting to see what will happen with that.

netAction, 2017-11-01:
Locking the screen works when a video plays: https://github.com/richtr/NoSleep.js

mxtommy, 2017-11-06:
fullscreen and nosleep done in commit 40d122b40c83718d6ada47f1fa113e70c25c59da ! Published to NPM as version 0.1.3

For history API, for now it's on the wishlist. I added it to readme roadmap. (will show up next commit)

closed 2017-11-06 by mxtommy