mxtommy/Kip #8: Extra /signalk/  in server configuration

tkurki, 2017-12-29:
As the SK specification sets /signalk/  as the standard path I think it would be easier and more strightforward to configure Kip with just protocol + hostname: `http://demo.signalk.org` as opposed to `http://demo.signalk.org/signalk`.

mxtommy, 2018-01-03:
Personally I like that it's fully supplied as it gives flexibility, though you are right in that it should never change... I'll see if there's a way to have both (default to /signalk if not specified, but if something is specified use that)

tkurki, 2018-01-03:
Easy enough to probe the suppliead url as is and then append /signalk if there is no proper sk response there.

tkurki, 2018-01-03:
Personally I think we should strive for doing things the spec way so as not to introduce extra variations. I’ve not been perfect that way, but I aim better...

mxtommy, 2018-01-16:
Done! if there's not already there it adds it.

closed 2018-01-16 by mxtommy