mxtommy/Kip #10: TTL Timer on data.

mxtommy, 2018-01-21:
Add a timer to the gauges so that they'd indicate lack of data if they've stopped updating.

Need to see how to handle non-changing information.

mairas, 2018-01-21:
Oops, indeed, this was what I was referring to! :-)

JonyIvy, 2023-03-29:
Is there a solution for this yet? Same problem here with engine data (e.g. RPM). When engine is turned off and there is no more data coming, the gauges should switch to a preset default value or deactivate themselves visually.

JonyIvy, 2023-04-03:
In case someone has the same need, I solved this with a node-red flow that sends an RPM message with value null if there was no RPM message since 3 seconds:
`[
    {
        "id": "0c315078475f2c47",
        "type": "signalk-input-handler",
        "z": "cb32c58b39aa72a8",
        "name": "revolutions",
        "context": "vessels.self",
        "path": "propulsion.main.revolutions",
        "source": "",
        "x": 100,
        "y": 860,
        "wires": [
            [
                "c323c3351add9d5c",
                "a55f80daa2947bca"
            ]
        ]
    },
    {
        "id": "c323c3351add9d5c",
        "type": "signalk-input-handler-next",
        "z": "cb32c58b39aa72a8",
        "name": "",
        "x": 640,
        "y": 860,
        "wires": []
    },
    {
        "id": "5fcf1962f8e4f642",
        "type": "trigger",
        "z": "cb32c58b39aa72a8",
        "name": "timeout send null",
        "op1": "",
        "op2": "",
        "op1type": "nul",
        "op2type": "pay",
        "duration": "3",
        "extend": true,
        "overrideDelay": false,
        "units": "s",
        "reset": "",
        "bytopic": "all",
        "topic": "topic",
        "outputs": 1,
        "x": 450,
        "y": 980,
        "wires": [
            [
                "3949d620cfd8dcfa"
            ]
        ]
    },
    {
        "id": "a55f80daa2947bca",
        "type": "function",
        "z": "cb32c58b39aa72a8",
        "name": "build null msg from original",
        "func": "msg.payload = null;\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 260,
        "y": 920,
        "wires": [
            [
                "5fcf1962f8e4f642"
            ]
        ]
    },
    {
        "id": "3949d620cfd8dcfa",
        "type": "signalk-input-handler-next",
        "z": "cb32c58b39aa72a8",
        "name": "",
        "x": 640,
        "y": 920,
        "wires": []
    }
]
`

godind, 2023-09-23:
Feature planned with Metadata Zones management feature. ETA unknown as it requires SK improved Metadata editing support but work has started.

ignamendi21, 2023-10-30:
please help..
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
at 5974 (main.js:1:64836)
at n (runtime.js:1:128)
at main.js:1:64914
at f (runtime.js:1:698)
at main.js:1:49

Thanks!!! : )

commit 78fbb750 (2023-12-05): TTL Timer on data.
Fixes #10 UI updates
PR #242 (2023-12-05): Godind/issue10
Fixes #10 Support Widget Path Value TTL
closed 2023-12-05 by godind