mxtommy/Kip #19: Deployment details

rob42, 2018-07-07:
Nice work, really like the project :-)
Ive been installing via signalk-java using the new artemis server.

1) Install is fine (signalk-java does a git clone). Signalk-java then does an 'npm install', and 'npm build', I then needed a custom 'ng build' from the command line.

In the interests of auto-installs, it would be good if it all worked with std calls.

2) In signalk-java apps are installed in signalk-static/{app-name}/, where signalk-static is the root of the web server, eg '/' Hence the app is expected to have an index.html in the root dir which launches the app when http//myserver:8080/Kip is executed.

In Kip I need to launch /Kip/public/index.html. Then it fails as the <base> tag is '/'
So:
  Please add an index.html in the root, which redirects to public/
  Adjust <base> to work as <base href="/Kip/public/"> That may need some thought as Im not sure how other projects deploy?

rob42, 2018-07-07:
On retest the 'npm run build' called the 'ng build' so 1) above is closed.

mxtommy, 2018-07-07:
Adding an index that redirects wouldn't be that hard, however the more complicated thing would be setting Angular's "base-href" tag correctly. in this case I imagine it would need to be "/kip/public". You can set it with a custom build command (see https://github.com/mxtommy/Kip/blob/master/package.json#L21 for how I build it for NPM which the node server uses)  

I'm not quite sure how I could automate that though. Angular environment expects you to build locally and deploy the built artifacts. With npm package for node server I build locally manually, and can set the href. (artifacts end up in /public) and node server serves from public. 

I'm open to ideas if you have any! :)

closed 2019-07-17 by mxtommy