mxtommy/Kip #24: Feature request: Provide Vertical time-series chart for TWD & TWS

zubenubi, 2018-11-27:
A vertical strip chart for TWD and TWS paths is a hugely useful feature for competitive sailors to help their tactical decisionmaking. Example attached
[image]

RhumbRunner, 2020-05-23:
I also would second this request. I would envision that the graph would use Min/Max values for determining the horizontal values. Also should the Vertical values be minutes with a setting to show how far into the past for the data history or should it be dynamic with a fixed vertical height and the maximum value be determined by the height of the widget?

mxtommy, 2021-10-13:
Done thanks to ahm4711!