mxtommy/Kip #26: Demo is not working after last update 0.1.9

sailoog, 2019-04-17:
running on signalk k server 1.14.0
Also lost prior configuration and wouldn't run prior config when pasted it in.

mxtommy, 2019-07-17:
Fixed in 0.1.10 :)

Note prior config still wont work when pasted in. The config version is different. (means there is different structure/data in the config, if it tried to load it app would likely crash). Do you know how other projects handle this type of problem? :)

closed 2019-07-17 by mxtommy

sailoog, 2019-07-17:
Thanks! I will check ASAP. I want to add KIP by default in the new openplotter version.

I am afraid the only way of handle that problem is marking you version as 0.2.0. 
Normally versioning works like this:
<changes make this version no compatible with the old ones>.<changes require some dependencies to be updated or add new features>.<changes just contain bug fixes>
So, according to that logic you should mark it as 1.0.0 :)