mxtommy/Kip #30: Feature Request: Add ability to save default config on server.

NAHANNIV, 2019-12-05:
I think it would be useful if you could have a way to store a default configuration on the server.
Then each new device would not have to set everything up from a blank page.
Especially useful when the configurations become complex.

mxtommy, 2019-12-06:
Indeed this would be useful. I think they had started a server plugin to store data, not sure what happened with it. I'll try and find some time. :)

skipperearly, 2020-02-24:
How about extending the "save" function in the config tab with a "save as" and "Load" to the localhost, basically doing the cut/paste for the user with a text file on the server. 
Just my 2 cent.

Jamos1988, 2020-04-05:
Looking for this feature as well, I am running on multiple devices / browsers potentially, would be a great feature! So, +1 here!

mxtommy, 2020-04-18:
Done in da21dcda240683d6a1360b31775d4f861cfd402e ! Will try and push new version to NPM soon!

Note requires Node server version 1.27.0 or greater

It doesn't load a saved config by default, but not much work to load it. 

To save to server you also need to have a R/W Token. Request it in settings under Signal K, then you have to approve it in the Node server admin page. In order to save to global scope (to use in other places) you also have to set that device to Admin permissions in Security. The user scope is limited to the token you are using.

closed 2020-04-18 by mxtommy

VibroAxe, 2022-08-06:
@mxtommy It would be really useful if kip could load a "default" config (perhaps one named "default" if it exists) in addition to this save / load. As you say above, not a huge ammount of work for the user, but more complexity than required I feel. Would you feel this would be a new FR or a reopen of an old?