mxtommy/Kip #33: Feature request: Drag and move of individual frames

skipperearly, 2020-02-24:
I found out the possibility to drag the frames and like it a lot! Thanks for this - but with eating the appetite  grows :) - so now I would like to propose a drag and place for individual frames - or maybe just a drag and replace/swap the source- with the target frame.

mxtommy, 2020-03-17:
To be honest, that would require a complete re-write of the layout stuff. A bit over my head right at the moment but I'll leave the issue open as it's not a bad idea :)

petsicola, 2024-01-16:
Antoher idea may be copy a the path and the other configuration of a frame into another frame. like copy and paste. or create a list of all boxes with it names and when you create another tell wich configuration you want to copy.

godind, 2024-01-16:
The challenge is widget don't have the same setup so it would be a full replacement and even so, not so simple to build.

That said, pages and widget setup is not KIP's primary use case. Visualisation is. KIP is a MFD replacement and meant to be used at the helm.

Once the setup is done, maybe a couple of hours tops, modifications are infrequent. On my boat, I don't find myself constantly changing the layout on my Raymarine chartplotter.

With this in mind I think this would be a nice feature but not critical to the product's success.  There are other solutions such as Grafana to complement the full solution.

adam6806, 2024-03-31:
I came here to request this feature. In the project description it says "Layout and configuration should be both easy to operate and flexible". If you setup a whole page of lots of widgets and decide you want to reorder them then it's very painful to do which is not easy to operate or flexible. This is a major gap. I think your point about once setup is done which should only take a couple hours with infrequent modifications is not the typical user and especially not the beginner. I've spent days fiddling and moving things around and seeing what the different pieces of data look like. I'm still not remotely done. If I could shift widgets around this would cut the time I've already spent at least in half and I would be happier with the results. So personally, I think this is more than a nice to have and more a critical piece. I may take a look at the code and see if there's something I can put in there even if it's rudimentary to move something from one place to another. This one thing is the only thing pushing me to use the default data display tool in Signalk because you can move things around after adding them. Everything else about Kip is better but this is so critical it might keep me from using it.

godind, 2024-06-01:
I'm working on something. It's a big rewrite and I hope I don't have make this a breaking change. 

I'll probably make this part of v3.

godind, 2025-03-09:
It will be a breaking change and released in KIP v3. Works great!

closed 2025-04-15 by godind