mxtommy/Kip #34: Feature request: Gain control over font size

skipperearly, 2020-03-13:
Dont know if this project is still alife - anyhow I try :)
When configuring text fields (but also for numeric fields) no control is given to the font size. 
The result for e.g. display system date/time or coordinates is therefore not readable.
[image]

mxtommy, 2020-03-17:
Hi! I try and work on the project when I find motivation :) 

For numeric fields the font size is automatic. Basically it starts with small font size and increases the font size until whatever is being displayed wouldn't fit in the window. I'm not sure it's useful to adjust it further? 

For Text fields, I must say I never really had them so didn't do the same trick. I could though. Let me see :)

mxtommy, 2020-04-18:
Done in f133790049616dac34995c5efc42f62e6fc28a4a :)

closed 2020-04-18 by mxtommy