mxtommy/Kip #40: polifill pour eink kobo browser

circus37, 2020-03-21:
hello i test kip on a kobo browser is very old but whith help off kobo developer they talk to me its possible with polifill i install all the polifills of polifills.ts but it dont work i think it s the localstorage fonction not work in kobo browser have you an idea thinks for your work great aplication

mxtommy, 2020-04-07:
Hi, Unfortunately I don't have a Kobo to test with, it makes developing for it not really feasble for me. :( If all that is missing is localstorage, Maybe I can see about it if I can get saving config to the server working (see issue #30 ) I cannot promise anything though.

circus37, 2020-04-08:
think you for your response i test the polifill include in the code but i don t know if they are active on the kobo i think its dificult beucause it very little weigt browser whithout lot of missed fonction but for example sailgauge work but the color isn t good lot of grey kip is better for personalitasion digitalyacht it more difficult because my goal it olny a gps repeter thinks a lot for your work

closed 2020-04-08 by circus37