mxtommy/Kip #41: Submit my contribution - New Gauges and Theme

godind, 2020-04-06:
Hi mxtommy,

I want to build a autopilot widget and since I know little about most of this tech, I started with a theme, then added new linear and radial gauge using ng-canvas-gauges. They all support your themes and config panels. I think I've done a pretty good job for a first timer. I'm not 100% done tuning colors schemes and such to make the gauges look really good but it should take me a few days.

How can I submit this to the project? Sorry but I'm new to this...

Note the Unit label in those screenshots are off: their is no power instrument data in the simulation stream so used paths to different streams just to display something.
 
[image]
[image]
[image]

Thanks

mxtommy, 2020-04-07:
Hi!

That looks awesome!

For sure I would be happy to merge your contributions! 

Contributions would be easiest by pull request. You mentioned you're new at this. If you think you will contribute to other projects in the future it can't hurt learning how it's done. You can see here:
[https://github.com/firstcontributions/first-contributions](https://github.com/firstcontributions/first-contributions)

I'd be happy to help you if you have any questions with that :)

Alternatively if you would rather, I suppose you could send me a zip file with you changes and I can try merging them manually :)

Thanks!

mxtommy, 2020-04-17:
PR Merged! Will push new npm version out this weekend!

closed 2020-04-17 by mxtommy