mxtommy/Kip #43: Dynamics theme selection and widgets

godind, 2020-04-12:
Hi @mxtommy,

When changing theme, some gauge elements don't automatically refresh their styles. You can see this with widget-numeric: change the theme and Title and Units texts don't update fillStyle color, but Value and MinMax do. I have similar issue with ng-gauges I added.

Reloading the browser page, resizing widget zones or browser window, or changing gauge page will refresh styles with new theme value.

**Investigation**
The issue is present for all styling application that comes from typescript. Reason is material components work nice with material themes, but elementRef and getComputedStyle not so much - at least not automatically.

**Issue**
All styling changes applied in ts, as opposed to in templates, are not applied automatically on theme change. Some event based mechanism is needed to trigger custom styling code updates such as getComputedStyle, or whatever needs to be done in the widget.

Example: the only reason widget-numeric Value and MinMax texts updates instantly on theme change, is because they are drawn based on Subscription events which is constantly being raised. Title and Units texts are only drawn in `resizeWidget()`. This is true for any widget that sets styles in code. Setting styles in code is I think essential or the only way to style in many cases.

**Proposed Fix**
We need a theme change event in widgets. This will allow us to trigger ts related styling code.

I've found that AppSettingsService exposes Observable `this.AppSettingsService.getThemeNameAsO()`, `this.AppSettingsService.themName` and a bunch more theme related stuff. I am playing with  `this.AppSettingsService.getThemeNameAsO().subscribe()` and still have not figure out how events and such work in Angular, but I think I am on the right track.

@mxtommy do you think this is the right approach?

godind, 2020-04-13:
I've done a bit of research and even with an observable subscription it's no use because at the time the event is triggered in the component the AppComponent is also swapping the theme. The widget observer is calling `getComputedStyle` at the same and the new theme styles have not been processed by the View yet. I've tried and it does refresh, but always with the value of the last selected theme.

I think this approach could be good but we would need to change the Observabe to a Subject and  use multicast. This way we could have multiple widgets in the same page that don't individually create their own instance, and subcription is easier to use. See [ https://medium.com/grensesnittet/dynamic-themes-in-angular-material-b6dc0c88dfd7](url) 

Or maybe this is just beyond my amateur skills!

godind, 2020-04-13:
found a hack by using a 50ms setTimeout on the event. Works! Let me push my changes and you can look at it. Next step would be to use Subject.

mxtommy, 2020-04-17:
Great work :) Thanks for it! PR is merged :)

closed 2020-04-17 by mxtommy