mxtommy/Kip #47: SignalK-Full.service data type error

godind, 2020-04-19:
Hi @mxtommy,

I've started using streams from other than demo site to test with Electrical and other types of telemetry. The following error pop's up but I'm not how to fix it as I don't yet fully understand the Kip SignalK framework.

> TypeError: Cannot use 'in' operator to search for 'timestamp' in null
> Setting self to: vessels.urn:mrn:signalk:uuid:6a59e3cd-7ce2-45a7-a156-0270e9be48d7
> Error: <path> attribute d: Expected number, "M 250,250 L NaN,NaN A 160,16…".
> Error: <path> attribute d: Expected number, "M 250,250 L NaN,NaN A 160,16…".

It comes from src\app\signalk-full.service.ts in `findKey()` function. Exception is thrown at line 34 `else if ('timestamp' in data) { // is it a timestamped value?`. 

The reason is the `data` object is `Null`. `data` is actually http `response.body` and it can be `Null` if no body was returned, which is not required and is a valid http operation. I looked at the `response.header` and all is fine: 200, OK, etc...

From what I read, you can only use `in` on singletons, objects, parameters or any. Obviously Null is a problem but I think the logic needs to be looked at as a Null HTTP response.body is something that we need to expect in some cases. Here is the `data` in question attached as JSON maybe it will ring a bell!

[data.json.txt](https://github.com/mxtommy/Kip/files/4499965/data.json.txt)

mxtommy, 2020-04-19:
Already patched here:
https://github.com/mxtommy/Kip/blob/master/src/app/signalk-full.service.ts#L26

It's a recursive function to walk the full json tree, I hit the same thing with the notification branch.

What server are you using? There seems to be a lot of bad data in there?  Like right at the start I see design.length has both value and values keys, not sure that is valid

godind, 2020-04-19:
it's a file from sbender. It has auto-pilot, some Victron and battery info.

get from [(https://www.dropbox.com/s/wesky0kgqignrmh/signalk-rawdata.log.to.langford.bz2?dl=0)]

use Multiplex as fromat

godind, 2020-05-03:
Fixed

closed 2020-05-03 by godind