mxtommy/Kip #57: Feature Request: Display Time / Timers

RhumbRunner, 2020-05-23:
Paths such as navigation.racing.timeToStart should be able to be displayed. SI is seconds but should be able to be Displayed as M:S.

Paths with an elapsed time navigation.trip.lastReset which I am using to hold an elapsed time from a the reset point. This I am storing in Seconds but should display as M:S or H:M or H:M:S
This would also be useful for values such as engine hours or just timers for Predicted Log navigation.

Other Time Paths should also be able to be displayed in a H:M:S (AM/PM) for a 12HR format or a HHMM:SS for a 24 hr clock.

godind, 2020-05-23:
I guess you are racing! Could you point us to a concept or similar display? Are you simply thinking of some kind of watch/clock faceplate?

RhumbRunner, 2020-05-24:
I was thinking just a simple readout of the time just like other numerical widgets. I’m starting on a python plugin that will allow you to start/stop and sync based on Boolean pulses from KIP. 

This picture is from the iPhone/iPad app that I usually use.  This app also allows you to “ping” the ends of the starting line and gives you a distance to that line. That’s a little ambitious for my programming skills.

Thanks for looking at this,

Chuck Bowers
Rhumb Runner J/29
(619)905-1398 Cell

godind, 2020-05-25:
Can’t see any pictures. Sorry but I’m still not clear, you want to display time coming from a SignalK time Source? Like a GPS?

With a gps their are lots of possibilities to create race type gauges.

What do you race? I raced J22 959 in the Canadian and US class for some years, and other PHRF class of course. Managing 2 boats and leaving often for NA and World regattas was too much with new bornes so I kept the cruiser!

mxtommy, 2020-06-07:
Added HH:MM:SS unit type in 06670a113ef1932071231e0596568d64da59cb65 :) Will be release in next version!

closed 2020-06-07 by mxtommy