mxtommy/Kip #61: version 1.1.2 - Stuck in demo mode

RhumbRunner, 2020-06-14:
Just updated to 1.1.2 I have attempted to load the previous saved settings. Webpage says that it has loaded and It immediately displays the Welcome to KIP. Going back into settings I see that the token has an X. I have gone into the server and deleted and re-requested / issued a token 3 times now and as soon as I load either the default or my saved configuration. It returns to the Welcom to KIP screen and the token will show an X.

Not sure where to go from here. Will post this on Slack #Plugins as well...

-- Chuck

mxtommy, 2020-06-14:
Replied on slack:

Hi @Rhumb Runner Sorry about that, I'm a bit of a noob developer so haven't found a nice way to deal with updating the way kip stores the config. Basically when I change config in some incompatible way, I set a flag to invalidate older style configs, otherwise kip may load a config that it no longer makes any sense to it. So when you try and load an old config it re-clears the config (which returns you to the welcome screen)

basically you will need to redo the config :disappointed: Sorry!

closed 2020-06-14 by mxtommy

godind, 2020-06-14:
@RhumbRunner @mxtommy are you using sever config storage or local config?

As mxtommy pointed out, when config storage structure changes, the app:
1- flushes all config, start blank
2- blank config has a default screen that offers users to load Demo config.

From their you have 2 options, either you load demo config, and adjust to your liking, or you don’t load demo and simply co figure url, add pages, gauges.  Kip automatically saves all these changes for you as you work.

But you seem to maybe have a token request issue. Note that any time you hit Clear Config or Load Demo, Kip will flush you settings, including token.