mxtommy/Kip #62: Feature Request: Change Major ticks on Radial Compasses to Degrees

RhumbRunner, 2020-06-23:
Have the ability to change the major Ticks on the radial compasses to 0,45,90, etc. in place of N,NE,E, etc. I can see where it is in the widget-gauge-ng-radial.component.ts file but wouldn't know how to begin to make it a check box or something as part of the setup when you create the widget.

Thanks,

Chuck

godind, 2020-06-23:
Hi Chuck

All widget options are in the modal-widget control. They are tied to panels. Take a look and let me know if you have questions.

I noticed the marine compass looks broken. Compass should have a fixed needle to 0 and only the faceplate turns. Other compass types the meddle turns.

Do you see that problem on your end?

godind, 2020-06-23:
Just better read your question!

They way it works at a high level is: you need to bind the modal panel option to a widget config option. Say Cardinal Points or Degrees display checkbox, then create a gauge major-tick conditional setting based on the Widget options.

The widget options properties is what gets saved in the config. Find one existing property you could reuse to store this. Look up the widget option interface definition and you will see all possible property you could use to store the Boolean value. You cloud add one also but make it optional using « ? ». Else all widgets will require this property.

I’m not in front of a computer so working from memory. Let me know if you have questions.

mxtommy, 2021-06-03:
Added numeric ticks in dcc253a03ff8a7c484bfc8e5ade35163fb776728

Note there's a bug in canvas-gauges which is preventing the needle from being fixed at top. I opened a PR ( https://github.com/Mikhus/canvas-gauges/pull/240 ) but not sure how active the repo is, last commit was a long time ago. Fingers crossed!

closed 2021-06-03 by mxtommy