mxtommy/Kip #70: Text orientation for apparent/true wind

deadeye12, 2021-04-01:
Great work on this!   Not critical, and not sure why it bothers me :-)   but would it be possible to have an option on the compass/wind display show the text for AWA and TWA right side up depending on where the wind was coming from?  For example if you are close hauled, the AWA number is upside down, on a run the true wind angle is upside down.  Could we have the option to have Them flip orientation when it crosses the 'southern' or 'northern' half of the rose, thus being more readable at a rapid glance?  -Thx!

godind, 2022-11-15:
Not sure I understand. Can you share a screenshot as a concept?

godind, 2023-11-07:
Just took another look and I think I got it!

Actually working Wind Widget. I'll give it a try!

closed 2023-11-11 by godind