mxtommy/Kip #82: Feature Request: Windgauge - indicator bearing to next waypoint

Sjoerd02, 2021-05-30:
To me it would be usefull to have an indicator to the next waypoint in the windgauge.
By having that you can see directly if you can reach it based on the laylines.
On non tacking course you can obviously see if you are heading in right direction.

Sjoerd02, 2021-05-30:
Thinking over the comment on non-tacking course, you might want to have an extra indicator for COG to see if you are really heading to you waypoint

davidsanner, 2022-01-17:
Windgauge is a great instrument.  One thing I'd like to see is the ability to diplay data (numerical values) in the space in the corners around the gauge. eg. VMG, STW, or even wind speed (the numbers on the gauge are hard to see on my tablet).

godind, 2022-08-10:
> Thinking over the comment on non-tacking course, you might want to have an extra indicator for COG to see if you are really heading to you waypoint

I think that navigation relayed info should be handled by the AP gauge, or maybe a new gauge is required. Wind gauge is really all about wind speed, wind angle and tacking. VMG is one gauge you can add to see wind angles vs course performance value.

Techstyleuk, 2023-07-07:
I really like the B&G SailSteer view and they indicate True wind, Apparent wind, Bearing to Waypoint, heading, COG, Laylines and current direction and speed.  I think KIP has most of this with exception of bearing, COG and current which I think would be a great addition.  I also agree that using the four corners would be great!!

I love this plugin and use it all the time!  thanks for all the work that has been done so far!!

godind, 2023-11-22:
We will add next point to the wind widget in the next release. Have a look at the new improved UI in the lastest version!

petsicola, 2023-11-27:
I think that people say something like this.
Number 1 is a graphical current direcction and current speed
Number 2 is the next point. i put here an image. 
I´m sure that you know this B&G Dashboard... but I preffer your new wind gauge theme ;)

[image]

godind, 2023-12-09:
Not exactly the same UI but implemented.

closed 2023-12-09 by godind