mxtommy/Kip #86: Problem

joma42, 2021-09-01:
I am useing mxtommy/kip for a wihle and I am quite happy, But since a while i have a problem displaying linear gauges propper and this only on one page. I created this page new after updating to 1.2.1. 
[image]
I have no idea if this can be related. I am grateful for any advice

mxtommy, 2021-09-06:
Are you using Firefox? Is so might be related to #84 If so not much I can do unfortunately until it's fixed upstream. You can try Chrome though

joma42, 2021-09-08:
Thaks for your help. 
Using Crome solved the problem of "pixeling" the gauges. 
Unfortunately, it did not solve the problem that the gauges are not updated and the numerical values are not displayed in small fields.

[image]

After refreshing the page manually all gauges show the correct values but the display is still not updated automatically. When I switch back and forth between pages (inside KIP) all gauges show zero again. This happens with gauges of all types.
The problem with the numeric values not being displayed seems to be a scaling error. If I make the box big enough the values are displayed perfectly.
It would be great if you also have a solution for this.
Thanks in advance

godind, 2022-11-22:
Can you check again with the latest Kip version please and tell us what hardware, OS and browser version you are using.  We cannot recreate this bug and no one else has reported a similar issue. Thank you

godind, 2023-09-23:
No response to further investigation. Closing issue as this is the only report is such a problem.

closed 2023-09-23 by godind