mxtommy/Kip #96: Feature request / Idea AIS Radar view

skipperearly, 2021-11-23:
would be great to have a page or object  : AIS radar view 
the objective would be to have a display showning in any web browser on any devise a colision information of AIS targets in a radar like view.

theowlpo, 2022-03-02:
It would be great. Similar to the plugin used in OpenCPN. Could be as a start a simple radar display ownship in the middle and then some lines of bearing to the AIS targets around.
Thank you

kagouraki, 2022-06-16:
Yes. Ais must be definitely supported in kip. A simple radar will do. I think is the only thing missing right now

slurcott, 2022-11-11:
yes, this feature request has my vote

kagouraki, 2023-02-04:
Do we have some progress on this ?

godind, 2023-02-08:
It is a feature that we are interested to add but it is a significant task. We are, with our available bandwidth, working first on base features such as configuration sharing and metadata support; enhancing auto gauge configuration.

Have a nice day!

ignamendi21, 2023-10-30:
please help..
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
at 5974 (main.js:1:64836)
at n (runtime.js:1:128)
at main.js:1:64914
at f (runtime.js:1:698)
at main.js:1:49

Thanks!!! : )

godind, 2024-03-19:
Some simple SK plugin that could be used: https://www.npmjs.com/package/collision-detector

godind, 2026-01-02:
The component must:
	•	Use polar coordinates (bearing θ, distance r)
	•	Have range rings (e.g. 1 NM, 3 NM, 6 NM)
	•	Show heading line / own-ship center
	•	Plot multiple moving targets
	•	Support click / tap interaction
	•	Be high-performance for frequent updates
	•	Integrate cleanly with Angular (19) and change detection

Use Canvas-Based Option: Konva.js
	•	Canvas performance
	•	Built-in hit detection
	•	Easier than raw Canvas
	•	Cleaner than Pixi for UI-heavy apps

Pros
	•	Good interaction support
	•	Works well inside Angular
	•	Faster than SVG for medium loads

Cons
	•	Less flexible than D3 for polar math
	•	Styling radar grids is more manual

godind, 2026-01-20:
@kagouraki  @theowlpo  @skipperearly  @slurcott

Hi All,
I'm curious if we still need an AIS radar view. I think in most cases OpenCPN or Freeboard-SK is running in split mode with KIP, right? Isn't looking at AIS information much more valuable when visible over a chart in relation to your vessel and it's surrounding? Maybe the usefulness of a basic radar view is minimal. Maybe it's just a nice to have?

What do you think? Share some usecases you see where it would have value.

Thanks