mxtommy/Kip #113: Units of Direction [0,360] instead of angles [-180,180]

RhumbRunner, 2022-04-12:
The derived data plugin calculations for Course DTG, XTE, BRG, etc (based on courseGreatCircle.nextPoint / previousPoint) creates negative values for the bearingTrue and bearingMagnetic which is the calculated angle from North (0 Deg True) to the next point. So to get the compass bearing to steer to the next point you would have to subtract the value from 360. Can a units conversion be made to change these angles [-180,180] to directions [0,360]. In SI radians this would be angles of [-3.1415,3.1415] and directions of [0,6.2831] roughy. (pi being what it is)

godind, 2022-08-16:
Can you provide steps to configure such data and recreate this problem please?

RhumbRunner, 2022-10-02:
Ended up modifying the derived data calculations to put out the desired [0,360] data. It's probably best addressed from that side of the house. I'll send my modified code to that repository for consideration.

closed 2022-11-18 by godind