mxtommy/Kip #130: I have to reconfigure the layout every time I open the page

hans-martijn, 2022-08-17:
As in the title: every time I open the page I have to reconfigure the layout. Sometimes everything has been reset, sometimes just the sensors have gone; I can't figure out what is really wrong. Shouldn't the settings be saved on the server somewhere?

KIP is added to my signalk server from the signalk installation on a dedicated RPI 4 4GB with RaspberryPiOS Bullseye.

godind, 2022-08-17:
What version are you running?

I bit of context first: Kip configuration is stored in the browser's local storage. Each browser app has its own individual local storage; Firefox Kip config is not share with Chromium. Same for Chrome and others...all separate.

Also browser config, as with most app config for that matter, is per logged-in users.

Make sure you don't have enabled security policies that reset all users settings (sometime called Sessions Settings or Session Data) after users logout/reboot the computer. Next check that the browser you are using for Kip has no security policies configured that would prevent using local storage or policies that reset local storage every session.

If this does not resolve the issue, look at the Browser console log using the browser Developer tools, edit a layout and see if it lists any error. That might help. 

You can save the Kip config in Signalk Server abd later reload it or load it from another device/Browser app this enabling config sharing and reuse but the process is manual.

hans-martijn, 2022-08-24:
I think in Safari things are automatically deleted, but in chromium it saves my session data, so at least on my boat I don't have to reconfigure every time I check.

Thanks for your help, it's really appreciated!

Hans

closed 2022-08-24 by hans-martijn