mxtommy/Kip #137: Issue: Ratio expressed as percentages the numeric limits don't apply

RhumbRunner, 2022-10-02:
When a ratio is displayed in a format other than as a Ratio 0-1 (value multiplied by 100) the minimum integer and decimal places don't apply.  I wish I had Screen shots...

Example:  Min Integer 1, Min Decimal 1 - Original path value 
performance.polarSpeedRatio | 1.102938346207304 | ratio 
Display format: % - Percentage value (value unchanged) = 1.102938346207304%
Display format: % - Percentage value (value multiplied by 100)  = 110.2938346207304%
Display format: Ratio 0-1 (value multiplied by 100) = 110.3

godind, 2022-11-18:
I found why this happens. The conversion library adds a % character any conversion other than Ration (original value) and that makes the value a string before it's actually formatted for display. We'd have to change the conversation function and remove the % in the display to keep it as numeric. It's actually a bad idea to change the data type as part of a conversion but it's sometimes useful for display purposes.

Not sure if there is an actual use case where we need to keep integer and decimal places for Ratios as percentages. Can you think of a use case that could help settle this? Do you only use it in the Numeric Display widget?

The challenge is the conversion function is available to all widgets. To be honest I'm not sure how people apply Ratio data to widgets. Let me know what you this.

`//   ratio
    'percent': function(v) { return (v * 100) + '%' },
    'percentraw': function(v) { return v + '%' },
    'ratio': function(v) { return v * 100 },`

RhumbRunner, 2022-11-19:
That makes sense now why it didn’t work right. The only place I was using it was to show the percentage of our speed compared to our polar estimate. Just the integer is all I’d and most people would probably need. The raw ratio (0-1) even after multiplying by 100 would give you lots of decimals. Having the % sign is nice but maybe ultimately unnecessary if it’s in the units portion of the widget. You could just use…ratio 'percent': function(v) { return int(v * 100) + '%' }the only thing would be is to have a note that it’s strictly the integer percentage. If someone wanted the decimals they could use the ratio *100 and then tailor the max decimals and such. It’s what I am currently doing. Thanks for looking at this…Chuck BowersRhumb Runner J/29(619)905-1398 Cell

godind, 2022-11-19:
I'll remove the conversions and add a conditional format to the widget's final value instead. Thanks for the idea and use case. Helpful!

We will be releasing a version that supports automatic configuration sharing so the layouts get displayed everywhere. It's been bugging me that when you use another device to access Kip you have to rebuild everything. Like moving from the pi to my phone and tablet. We are waiting on a SK release that should be eminent.

godind, 2022-11-20:
Fixed in next release.

godind, 2022-11-25:
fixed in #149

closed 2022-11-25 by godind