mxtommy/Kip #169: N2K Autopilot rudder gauge is backwards

chalentech, 2023-02-07:
The rudder angle indicator at the top of the N2K Autopilot gauge (Component) backwards (inverted) starboard should be +.

version 1.3.0

godind, 2023-02-08:
What is your AP model/setup? Works fine on my boat. Can you also include a screenshot?

chalentech, 2023-02-08:
[image]

[image]

P70r control head 
ev-1 course computer e70096
Raymarine Rotary Rudder Reference Transducer

godind, 2023-02-10:
I don't have access to hardware at the moment due to winter season here. It will take something to look into this but I will get to it.

Thanks for the information.

godind, 2023-12-22:
I've done some investigation and your rudder sensor, the rather the N2K device converting the sensor data, is sending inverted values. That's what I see when compared to other user's and mine.

I'll and an invert rudders option.

Of note: in you screenshot I noticed you have multiple devices emitting values for the same paths. This can potentially be an issue (not related to the problem at hand) if you do not select a source. Both sources will overwrite each other for that given path in KIP. You should either configure SK source priority or check that a source is selected in KIP and not source "default". The latest UI should prevent this problem.

Mastery972, 2024-06-08:
Hi guys!
I have exactly the same issue. My autopilot is a Simrad AC20 with a Simrad AP16 screen.
Rudder is ok on AP16 screen and also on my Raymarine a65 MFD.
Unfortunately, it appears inverted with N2k Autopilot. I use the latest version of the plugins with a Victron Cerbo GX. I was not able to find an option to display it correctly.
Best regards,
Y.

Mastery972, 2024-06-09:
> > Hi guys!
> > I have exactly the same issue. My autopilot is a Simrad AC20 with a Simrad AP16 screen.
> > Rudder is ok on AP16 screen and also on my Raymarine a65 MFD.
> > Unfortunately, it appears inverted with N2k Autopilot. I use the latest version of the plugins with a Victron Cerbo GX. I was not able to find an option to display it correctly.
> > Best regards,
> > Y.
> 
> Can you share the SK Data Browser path value please? This will help. I probably just need to add an invert rudder option to the widget but want to make sure.

Hi godind
Find some screenshots here.
[image]
[image]

Best regards,
Y.

godind, 2024-06-10:
Hi,

Thanks for this. Sorry but I should have asked for more details before!

Because I have no hardware to test this, can you share SK's published Rad values from your install at:

Full Port:
Full Staboard:

It's probably positive on port and negative on stb but the range will also help.

Thanks

Mastery972, 2024-06-10:
Hi godind
Find here some screenshots of what you asked for:
[image]

[image]
 
[image]

Best regards,
Y.

closed 2024-06-12 by godind

godind, 2024-06-12:
@Mastery972 @chalentech feature published in v2.11.0. Have fun!

Mastery972, 2024-06-12:
Thanks a lot!
It works well in latest version (2.10.1).

Best regards,
Y.

godind, 2024-06-13:
My pleasure.

PR #426 (2024-06-12): N2K Autopilot rudder gauge sensor emitting inverted values
Fixes #169
commit 5ee3dd6a (2024-06-12): N2K Autopilot rudder gauge sensor emitting inverted values (#426)

Fixes #169