mxtommy/Kip #170: Negative zone alarms triggering incorrectly

tvr256, 2023-02-17:
I have a depth gauge widget displaying depth below keel.

If I configure an Alarm zone with empty lower value and upper value set to 0, the alarm triggers with positive depth values. I would expect it only to trigger with negative values.

godind, 2023-11-11:
@tvr256 Strange because you should nor be able save this setup. Empty lower and upper of 0 is not accepted. Can you please double check. Thank you

tvr256, 2023-12-21:
I've double checked, and I can save empty lower and upper of 0 in 2.4.2 :

[image]

godind, 2023-12-22:
I'll take a look. It should account from 0 down to negative infinity but it's probably the opposite.

Does it work if you do from -10000 to 0?

PR #335 (2024-03-21): Negative zone alarms triggering incorrectly
Fixes #170
closed 2024-03-21 by godind