mxtommy/Kip #171: Historic DataSet widget - initial value incorrect and persisted forever

tvr256, 2023-03-06:
I'm trying to use the Historical DataSet widget to display changes in barometric pressure. I'm capturing pressure every 5 minutes for the last 8 hours:

[image]

There are 2 separate issues. Firstly the initial value is extremely low, which throws out the scale and makes it hard to see small changes. Secondly the initial values don't disappear after 8 hours, instead they seem to stay forever:

[image]

JonyIvy, 2023-03-07:
For this use case I'm embedding a Grafana dashboard, which shows a graph of the pressure (data comes from InfluxDB). This way you have better access to the data and can configure the graph just the way you like.

godind, 2023-11-21:
Maybe be caused by sensor null values. We will if a filter fixes the issue.

godind, 2023-11-26:
This should be fixed (null or undefined value in datasets) in the next release.

Thank you

closed 2023-11-26 by godind