mxtommy/Kip #172: Feature Request: Date formatting

JonyIvy, 2023-03-07:
Kip is such a great app already, thank you very much!

Today I created a page with sun and moon rise/set times. The data comes from signalk-derived-data and is an ISO string (eg. 2023-03-07T09:49:00.000Z).
It would be nice to have a DateTime formatter that can transform this string to just hour and minute and that in local time. Maybe something like this would work in units.service.ts:

```
"DateTime time": function(v) {
      const d = new Date(v);
      return d.getHours() + ':' + d.getMinutes();
    },
```

or 

```
"DateTime time": function(v) {
      const d = new Date(v);
      return d.toTimeString();
    },
```

slurcott, 2023-03-23:
this can be done in NodeRed

JonyIvy, 2023-08-22:
I've seen there is a PR for this (https://github.com/mxtommy/Kip/pull/148). It would do what I need, if in a bit different way. What's the status on this PR?

godind, 2023-09-25:
@JonyIvy this will be part of V2 release. Soon...!

closed 2023-09-25 by godind