mxtommy/Kip #174: State (boolean) Widget display not working

basil59, 2023-03-23:
. Using Version 1.3.0
. I have a boolean value generated by SensESP representing bilge pump state
. Verified that value is updating correctly in Signal K's data browser
. Created a State (boolean) Value widget for the value
. Widget displays as a yellow box with the label, but no state indication (e.g. on/off)
. When value changes from false to true, the yellow color gets sightly darker, but still no indication of state (as shown in attached screenshots)
. Tried this with Chrome and Edge with same results. 
. Signalk/instrumentpanel displays the value as expected

Thank you for all the effort you have put into this software!

[image]
[image]

JonyIvy, 2023-03-27:
Same use case here with a bilge water sensor. It sent true or false but that didn't show well in KIP.
My workaround was to change the output to 0 or 1 and use a linear gauge (min 0, max 1) which changes color more visibly.
Still the state/boolean gauge would be nicer to use, so +1 for the CR.

basil59, 2023-09-20:
Any updates on this issue?  any way I can help?

godind, 2023-09-23:
Hi. Are you on version 1.3.0? I had fix issues with this control but maybe not all!

Question: what do you find is a better usecase for Kip compared to Instrumentpanel or other instruments apps?

Also, are you able to run Kip from Master branch? I need people to beta test a major release that is blocking other releases and biz fixes.

Thanks

basil59, 2023-09-24:
Hi, Yes, I'm on version 1.3.0, installed via the SignalK appstore

As far as use cases, Kip has a more refined and polished look that I would choose as the (production quality) interface to my SignalK installation, in particular, the 3d-look of widgets and the wide range finish and texture choices.  Also, setup and customization of the widgets is consistent and straight forward.

I would be glad to help test any version, I just need a quick guide on how to build and deploy.

Thanks!

godind, 2023-09-24:
Thanks for the feedback! It's always good to keep track of why a product exists and where things should go next. I'll get to fixing the button as soon as I get this important version released.

Thanks for you offer to help! Do you have JavaScript coding experience or any kind of programming knowledge? If not it you take some effort to set you up and it would probably be easier if I just let you know when I release so you can be an early tester. Do you have regular access to you boat or Pi installation to deploy and test? All my hardware is on my boat and not readily available.

Thanks

basil59, 2023-09-25:
Thanks for the quick response, glad to know the button fix is on your list.
I have programming experience, but not in JavaScript.
I do have regular access to the boat, and a test system I can use at home.

godind, 2023-09-25:
I looked at it. The intent was that the color change was indication of the changing state of the boolean value. Maybe not clear the way it's rendured?

basil59, 2023-09-25:
This was my first interaction with the boolean state widget so I wasn't sure how it was intended to work.
Looking at the widget rendering (as seen in the screenshots in my report), it's very difficult to determine the state of the variable.  
In the absence of a graphical indication, maybe more contrast between the true and false colors (e.g. green/red) would be clearer? thanks!

godind, 2023-09-25:
Agreed. I've implemented this but without ever using it myself - no such device in my setup!

I'll improve on it.

I guess for Boolean values we need the button/switch widget to turn things on/off as we do now and another widget similar to an "active" light indicator.

Ideally I could also do a multi-light indicator widget but that would require much more (spare) time.

basil59, 2023-09-25:
Exactly.  It would be an active light indicator for things like Ignition, Bilge Pump, etc.
Thanks for all the effort you're putting into this.

godind, 2023-10-10:
Hi @basil59 @JonyIvy. I made a quick fix in v2 beta 5. Far from good looking but until I do a SVG version it will have to do. It's a light indicator with a label and button. I have no way of testing it! Please send feedback!

godind, 2024-01-13:
Have a look at the new v2.6 Boolean Switches widget.

Button and Light indicator coming soon. Needs feedback on this first design as the other will follow the same design.

Thank you

closed 2024-01-13 by godind

basil59, 2024-01-17:
Thanks for adding the switch panel widget, I like the idea a lot!
A minor observation, the slider switch control size is a little large compared to the switch text and the panel title text sizes.