mxtommy/Kip #175: Feature Request: default to full screen mode

slurcott, 2023-03-23:
fullscreen mode looks better and I always switch to it on restarts

gregsyoung, 2023-04-13:
not sure your hardware/software environment… however 
Im using an older “NUC” with windows 10, 
I have it configured to auto login, & auto start chrome in FULL screen mode  to the kip URL .

>… in effect, i simply apply power to the NUC, and a minute or so later i have KIP in Full screen … no interaction required except powering it on..

ignamendi21, 2023-10-30:
please help..
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
at 5974 (main.js:1:64836)
at n (runtime.js:1:128)
at main.js:1:64914
at f (runtime.js:1:698)
at main.js:1:49

Thanks!!! : )

godind, 2023-11-07:
> fullscreen mode looks better and I always switch to it on restarts

Good idea. Adding to the backlog 👍

godind, 2023-11-11:
I've had a good look and it's not possible to enter full screen automatically. The Browser, for security reasons, first requires a user interaction (mouse mouvement, click, touch, etc.) to enter full screen, play sounds, etc. This would be annoying if websites or Add services could do it! Makes sense to me but it's a shame. I even looked at faking those actions but it's not possible for those same reasons.

Closing request as not feasible...

commit aa9df21a (2023-11-11): Feature Request: default to full screen mode
Fixes #175
closed 2023-11-11 by godind