mxtommy/Kip #177: Gauges stay black on initial load

JonyIvy, 2023-04-04:
Sometimes on first load of KIP, the gauges stay dark, don't show color highlights. This happens on Chromium/RPi and Safari/MacOS. When I reload, the problem disappears. See attached screenshot.
[image]

It's unlikely to be related but in the console I also see 404s for some fonts:
.../@mxtommy/webfonts/fa-solid-900.woff2
.../@mxtommy/webfonts/fa-solid-900.ttf
.../@mxtommy/webfonts/fa-regular-400.woff2
.../@mxtommy/webfonts/fa-regular-400.ttf

godind, 2023-10-09:
I can't reproduce this problem. Do you still see it?

godind, 2023-11-07:
This is potentially fixed. Not sure as I could not replicate but fixed a styling issue in v2.0.1

Thanks

godind, 2023-11-17:
I've tried to replicate once more with the specified browsers with no success. There is a good chance it's fixed as we changed style cascading order for dynamic widgets instances.

I'll leave it open for you to confirm when you find time.

JonyIvy, 2023-11-21:
Hasn't happened any more, I'm closing the issue.