mxtommy/Kip #178: Wind gauge "jumping" through N (0 degrees)

yiannisc, 2023-04-05:
Hi,

When sailing due north, or very close, as the boat goes between 357 and 3 degrees, the moment it goes through 360/0 the Wind Gauge seems to be doing a full rotation it looks like it jumps. All the other times it's smooth. In contrast the Radial Gauge - Marine Compass works fine.

godind, 2023-04-05:
Can you share Kip and Browser specs that you are running please?

Thanks

yiannisc, 2023-04-05:
Hi,

Thanks for the quick response.

I run Kip 1.3.0 on OpenPlotter 3 64-bit.
Browser: Chrome Version 111.0.5563.147 (and 112.0.5615.50). Both Official Build 64-bit on Windows 11.
Also in Chrome 111.0.5563.116 on Android 11.

Many thanks,
Yiannis

godind, 2023-11-05:
For some reason a fix was applied to wind angle/speed indicators but not the face plate. I'll fix it.

Do you find that the gauge of useful? Something's not 100% right  and trying to figure out what to adjust. I feel like there is too many things moving around but it's not visually clear what it means. Do you agree?

#Primary visual indicators goals
- apparent wind angle to monitor jibe/tack proximity and maximum sail trim/speed
- Apparent wind speed to indicate appropriate sail configuration (reefs and such)
- how tight you are to Closed Hauled to optimize VMG

##Secondary
- True wind speed and angle
- Wind shifts

#Minimal widget config options

##Required for optimal Widget usage:
- taking angle to reflect vessel specific Closed Hauled start ranges down to 70 degrees where Close Reach starts

##Advanced options
- Could have wind strength to vessel specific Reef setup with a  visual indicator
- real-time Closed Hauled adjustment following a polar map

closed 2023-11-11 by godind