mxtommy/Kip #179: Jumping Pointer/Display Radial Gauge

DerTom1978, 2023-04-08:
Hi,

The pointer of the Radial Gauge jumps back and forth from 0 to the correct value.
The numerical value underneath seems to stay at the correct value.
How can I prevent the pointer from jumping?

I have uploaded a video. This shows the problem

BR Tom

https://user-images.githubusercontent.com/51379696/230718738-04ba20b9-f5b1-4886-885b-6f8aaf15b757.mp4

DerTom1978, 2023-04-28:
i also tried this later with another PC and with another browser.
The result is unfortunately always the same.
The indicator jumps back and forth.

JonyIvy, 2023-08-27:
Same problem here with my SOG gauge.

godind, 2023-10-08:
@DerTom1978 @JonyIvy can you explain how I could repeat the problem? I can recreate it.

Maybe share the following gauge config details:
Display

- Min Integer and Decimal places
- Min and Max value

Path

- Signal K Path
- Data Source
- Display Format

Setting option

godind, 2023-10-08:
@JonyIvy  @DerTom1978 Can you also please check the KIP Data Browser to see if that data jumps. Filter on the path you are using:

- check the value with the default unit, then try changing to another unit by clicking the units button. Maybe the problem is in the units conversion and not the gauge it's self...?

godind, 2023-10-09:
I think I fixed it in v2 - next release. linear and radial gauge should also be smoother.

closed 2023-10-09 by godind