mxtommy/Kip #182: Value corruption (occasional)

JonyIvy, 2023-08-23:
I'm displaying the battery state of charge on my instrument panel in a basic numeric value widget. The value comes from VEdirect-serial-USB and is a float / ratio. Usually the value is shown correctly, but sometimes KIP adds digits, as if it made some internal calculation with the value and ends up with a rounding problem:

Value in SK: 0.944, value shown in KIP: 94.39999999999999999%

Settings of the widget:
Minimum integer places: 1
Minimum of decimals: 0
Display format: % - Percentage value (value multiplied by 100)

[image]
[image]

godind, 2023-08-23:
In your exemple, is 8.944 received the capacity percentage - the state of charge is at 8.944% ?

JonyIvy, 2023-08-23:
The state of charge was 94.4%, represented in SK as a ratio value of 0.944.
KIP is then multiplying by 100 and doing this in Javascript gives 94.39999999999. Yeah!

JonyIvy, 2023-08-23:
One could use .toFixed(n) to produce a nice output whenever a ratio (float) is multiplied by 100 for showing a percentage. The amount of decimal places (n) could be determined by looking at the number of decimals of the original value (.944 for 94.4% => one decimal, or number of decimals - 2).

godind, 2023-08-26:
It should account for configured Int and Decimal display format from the widget configuration properties. Is it not doing so? If not please confirm and report a bug. Thanks

JonyIvy, 2023-08-26:
The decimal setting label reads "minimum decimals", so actually it's ok when it's showing 12 decimals (= more than "minimum"). Or is this supposed to set a fixed number of decimals?

godind, 2023-08-28:
It's supposed to be decimal places to display.

ignamendi21, 2023-10-30:
please help..
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
at 5974 (main.js:1:64836)
at n (runtime.js:1:128)
at main.js:1:64914
at f (runtime.js:1:698)
at main.js:1:49

Thanks!!! : )

godind, 2023-11-07:
Is it ok or is there a bug?

godind, 2023-11-15:
I'm trying to reproduce this problem but can't. I'm using self.tanks.fuel.0.currentLevel - also a ratio value but it works fine! Could it be that you get `null` values from the SK? You might see something in the in the console log that could help.

JonyIvy, 2023-11-21:
Looks like the problem is fixed in v2.2.2. For testing I manually set the path to the value of 0.944 in data fiddler and the widget shows 0.9 when decimals is set to 1, 0.94 when decimals is set to 2. Just like it should be. Great!

closed 2023-11-21 by JonyIvy