mxtommy/Kip #184: Time on the bottom of historical graphs takes up too much space

RichFind, 2023-09-04:
Great plug in, however,
The time at the bottom of graphs takes up too much space. Suggest abbreviating hours to h and min to m also not sure that ago is needed at all.

RichFind, 2023-09-04:
Maybe just the hh:mm:ss format could work. Then the text would not be needed.

godind, 2023-10-09:
What version of KIP are you on?

Not sure we will work much on this Widget. I don't that many people use it really. Those that what graph data use Grafana and pull the graph in KIP using the Embedded Webpage Widget. More more powerful!

ignamendi21, 2023-10-30:
please help..
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
at 5974 (main.js:1:64836)
at n (runtime.js:1:128)
at main.js:1:64914
at f (runtime.js:1:698)
at main.js:1:49

Thanks!!! : )

godind, 2024-03-18:
Just released a new widget. Have a look.

Closing this issue

closed 2024-03-18 by godind