mxtommy/Kip #195: HOW TO COMPILE a modification i the project??

ignamendi21, 2023-10-30:
How can I modify something and build? I have run "ng build" or "ng build --configuration=production" but I am getting the same error. Kip is not showing anything, just WHITE PAGE./ I click to "inspect" and show this error:

**main.js:1 Uncaught TypeError: Cannot read properties of undefined (reading 'production')
    at 5974 (main.js:1:64836)
    at n (runtime.js:1:128)
    at main.js:1:64914
    at f (runtime.js:1:698)
    at main.js:1:49**

Thanks!!! : )

godind, 2023-10-30:
If you simply want to run locally to code, from the project root folder (kip folder) run command `npm run-script start`. To build a NPM release package use `npm run-script build-npm`.

godind, 2023-10-30:
Please don't spam all users!

ignamendi21, 2023-10-31:
I am just a bit desesperated.... Thank you @godind :)

ignamendi21, 2023-10-31:
i keep facen same issueee....  :(

[image]

[image]

godind, 2023-10-31:
You are running it from the  Signal K server?

What are you trying to do? What step?

First question before talking about deployment is can you run and test in your development IDE?

Tell me what tools and OS you are using.

ignamendi21, 2023-10-31:
Okey.
I installed Kip from Signalk but I want to do some modifications to it. Like colors in themes etc...

As I saw this repo in github I downloaded all the project and replace all the content of **/.signalk/node_modules/@mxtommy/kip** with what there is currently here in **master**, https://github.com/mxtommy/Kip

When I did this replce everything is fine, now I have this new version of kip runninf from signalk .

Now I want to do the updates. So I started changing a small thing , like for example a **label** of some html file, and I wanted to build it so I know I can do updates to the project and deploy them...

However when I do the **npm run npm-buil** and I go again to the endpoint of kip ,now everythin is in white, I go to **inspect** and see the error I mentioned before..  :(

godind, 2023-11-02:
You can't publish to KIP master branch or work from the installed KIP folders on your SK server. It won't work and anytime you want to keep up with KIP releases you will loose everything. Have a look the the KIP read me Instructions.

To build on Projects like KIP you need to fork the project to your own github repo.

Then using a development software (an IDE) such as VS Code (free), connect to your repo and check out your master locally. Once done, create a working branch from your master branch, check out and work on it. Once you are happy, merge your working branch onto your master, then make a pull request to KIP repo. We will review and merge into the product. 

Use the

closed 2023-11-04 by godind