mxtommy/Kip #198: Latitude and Longitude format

sanrab, 2023-11-08:
Hi! Thank you for your work. KIP is great !
In the last release I can't anymore have latitude and longitude data with format DEG MIN.HHH (Position latitude in minutes and longitude in minutes). The field remain empty. With option Degrees it works.
Previously the fields were correct.

godind, 2023-11-08:
May I ask to facilitate reproduction what widget and path you are using?

sanrab, 2023-11-09:
The widget I use is Numerical data, both for latitude and longitude.
The labels are "lat" and "lon".
The Signal K Path are self.navigation.position.latitude and self.navigation.position.longitude.
The data source is default.
With Display Format "degrees" data are shown, but with Display Format "Position Latitude in minutes" and "Position Longitude in minutes" data disappear (same for "Position Latitude in seconds" and "Position Longitude in seconds").
If I choose "Angles degrees" data appear again.

Hardware: Raspberry 3B, U-blox GPS on /dev/ttyACM0

sanrab, 2023-11-09:
mxtommy/kip version 2.0.1
SignalK server version 2.3.0

godind, 2023-11-09:
Awesome. Thank you!

closed 2023-11-11 by godind

sanrab, 2023-11-12:
Thank you very much. With KIP version 2.1.2 everything works as expected.
Thank you.
Sandro

godind, 2023-11-13:
Awesome. V2.1.3 now has a configuration update feature. If you have a v6 version lying around…

commit 426ae84e (2023-11-11): Latitude and Longitude format
Fixes #198
PR #204 (2023-11-11): Latitude and Longitude format
Fixes #198