mxtommy/Kip #199: Moving average to signalk data before kip display

ignamendi21, 2023-11-08:
Hello, 
First of all, thank you for the help I received so I have been available, finally to run my own builds locally.  :)

I have KIP and signalk  to display data in my boat. However the data is very "discrete", I can see how the AWA is changing every second with too big steps. 35-43-32-39-33-43, etc... I know that this is what the sensor is actually measuring and that it is correct, however it is not very usefull to see such a big change in the values. I was wondering if there is  a way to apply some kind of **moving average** to the data, so the data I show with kip is the average taking into a count the las 5 values (for example)

godind, 2023-11-08:
Hi

Thanks for the feedback. V2.0.1 has a new feature to throttle data update. In this release all widgets only emit data every 500ms so it should be a bit smoother.

In the next release this configuration field will be exposed in the paths options tab. It is not at the moment.

Hope this helps

ignamendi21, 2023-11-09:
@godind thank you for.your awnser!. When may it be released?

godind, 2023-11-09:
Working on some bugs. Will get to it after. Hard to say since it's on the Night Shift 😉

closed 2023-11-09 by godind