mxtommy/Kip #209: how to change background color in widget numeric?

ignamendi21, 2023-11-13:
Thanks @godind , for all your answers.. : )

Just a quick question.. I want to use de moderTheme, however I would like the Divs to have a black background instead of the grey... I have looking many files and I dont find where and what to change to have black background.

[image]
Could you tell me which file do I have to look at to change the grey color to the #0000000 ??

godind, 2023-11-14:
To change color schema, use the provided themes. What are you trying to do exactly? Build a custom version of KIP?

godind, 2023-11-14:
To obtain help on coding practices, source code, etc. please do not use Issues. Those are meant to track product bugs and enhancement requests.

Instead use the kip Signalk-dev Slack channel. You will find many ressources to assist on all development questions regarding Signal K and related apps. I will close this issue. See you on Slack.

closed 2023-11-14 by godind

ignamendi21, 2023-11-14:
Yes, just what I explained in the picture aboved. I want to use the moder-Theme but with black background