mxtommy/Kip #212: Older configurations missing w/ upgrade to 2.1.3

davidsanner, 2023-11-15:
I updated to 2.13 but now I am unable to view any of my layouts/configurations.

The config files are still on the server but when I copy/load them to local storage I see a quick flash of the old format and then nothing.  Old configVersion looks to be 6 while new one is 9?

Do they need to be converted somehow?  Thx.

godind, 2023-11-15:
It should upgrade automatically so it sounds like a bug.

Are you using "Sign in" mode or not? If you are, kip will use config named "default" in the user storage scope on the server, not config from local browser storage. 

Can you send me your original v6 config file? You can find it in ~/.signalk/applicationData subfolders. Any file before 9.0.0.json will do. It's probably in a 1.0.0 file.

I'll try and debug.

Thanks for reporting the issue!

davidsanner, 2023-11-18:
I couldn't get it working on multiple devices (w/ both a global config & my user config) but now, w/ the latest KIP release (or was it a restored config file) it's working fine.   Good news either way.

Lets call that bug fixed!  I'll go ahead and close it.

I really appreciate your help with this.  Great plugin as well!

closed 2023-11-18 by davidsanner