mxtommy/Kip #218: How do I load configuation in 2.2.1

markante, 2023-11-18:
I just updated to 2.2.1 and have the issue that I can not find the way to load my global/server configuation with the new storage management. Is all my work I put in the config gone and I can not load it anymore ?

Thanks for your help

godind, 2023-11-18:
Hi. It should have upgraded the local browser config. Unless you had never connected to KIP with that device. Is this the case?

Good news is it's not gone!

Look for ~/.signalk/applicationData/global/kip/1.0.0.json. Make a backup.

Rename to 9.0.0.json and see if you can see it from KIP Configuration tab.

If you use Sign in. You have to copy global to "user / default".

If not using Sign in, copy to local storage.

markante, 2023-11-19:
Thank you very much. Will try to do it with the jason. Have a nice day. I really like kip

godind, 2023-11-19:
Me pleasure. Let me know if it works.

godind, 2023-11-21:
Just released 2.2.2 that fixes configuration upgrade issues. Try it out and provide feedback.

Thanks for your patience!

godind, 2023-11-22:
FYI. V2.2.2 was released and fixes the problem. Please try this version and report back. Thank you!

closed 2023-11-26 by godind