mxtommy/Kip #221: Text doesn't scale well in Boolean value widgets

EdKok, 2023-11-22:
[image]

EdKok, 2023-11-22:
The switch doesn't scale nicely either and could be made bettr looking like:
[image]
See: https://www.cssscript.com/switch-web-component-jellyswitch/

EdKok, 2023-11-22:
If there would just be one graphical item, on the left, and text on the right, the user would be able to size the boxes equally without distorting the text.

godind, 2023-11-23:
We will evolve this widget in the future. It was just a quick patch as we could not really see the button state before. The button widget is getting more popular with time😀

The challenge with font sizing is it need to adapt to the space available.

If you have many button widgets of the same size on one page, they will all have equal text area "space" available. Then if the the widget labels, the string of characters,are not of equal length for each widgets, it has to vary font size so it fits in that space, which results in uneven widget labels.

If the font size does not vary, the labels would overflow.

I don't see an easy answer for this problem. It a challenge with all widgets. Try the numeric widget with a short and long label.

Thanks for the feedback!

EdKok, 2023-11-24:
But what about my suggestion of using juste one graphic? Placed left, leaving room for text at the right. The user can resize the length (horizontal) till all panels are equally long and text has sufficient room to expand.

EdKok, 2023-11-24:
Rough example made in Paint:
[image]

godind, 2023-11-24:
That would be an improvement. But if the label was longer than the space provided, it would still have to overflow or resize the font to fit that space.

We will have to rebuild the widget  in SVG to have better control.

Thanks for the feedback. We will get to it!

godind, 2024-01-13:
See new Boolean Switches Widget

closed 2024-01-13 by godind