mxtommy/Kip #222: Radial gauge spinning

JonyIvy, 2023-11-22:
Sometimes my radial gauge which displays SOG starts spinning, the whole gauge, not the finger. After reload it's fine again.

[image]

JonyIvy, 2023-11-22:
[image]

godind, 2023-11-23:
Hum... that's probably a problem in one of the gauge framework KIP uses. I'll take a look but we might have to wait for an update from the publisher. Which in open source is not always very frequent.

Can you share browser and OS version info and test the with another browser to see if it has the same issue. I noticed Firefox seems to have more issues with this gauge library. 

Use a steelgauge widget instead. It offers the same functionality. That should fix the problem but it does not look the same.

JonyIvy, 2023-11-23:
Safari Version 17.1 (18616.2.9.11.9, 18616) on Mac
Safari on iPadOS 16.7 (20H19)

I could not reproduce this behaviour on Firefox or Chrome.

godind, 2023-11-23:
I've been running on MacOs Safari 16.5 for 12h with no issue. I'll upgrade to the latest and see if it shows up. I'm beginning to suspect that fidget maybe gets bad data that messes things up.

godind, 2023-11-23:
Are you using a stream file or it's live on your boat?

JonyIvy, 2023-11-24:
I'm using a stream file. It happens at seemingly random times in the stream. The only pattern I see is that the bug appears about every 12 minutes on iPad, 14 minutes on Mac. I tested this for more than an hour.

godind, 2023-12-05:
I can't reproduce and no other user has reported this, yet... so I think it might be related to data values. The latest 2.3 release has better filtering of null values.

Try and see if it fixes the problem.

godind, 2023-12-21:
Have you seen the problem in the latest release?

JonyIvy, 2023-12-21:
Nope, it has been working fine since one hour now. :)

JonyIvy, 2023-12-21:
Yes it is still happening. I will test with SK simulator data now to make sure it's not my logged data.

godind, 2024-01-14:
There is a fix on the Canvas-Gauge package to fix this. It's been pending 3 years in PR... I sent an email to the owner. Maybe we'll need to switch gauge library... Thant would be a lot of work :(

godind, 2024-02-27:
still have this issue? I've never been able to reproduce so far. Not sure what the problem is and no one reported a similar problem.

JonyIvy, 2024-03-03:
I have seen it today with the latest version of KIP but still with my old data log.

godind, 2024-03-04:
It's probably related to the computer it's running on. Check for OS, browser and node updates.

godind, 2024-05-28:
This should be fixed in V2.10. Please have a look and report back. Thank you for your patience!

closed 2024-05-28 by godind