mxtommy/Kip #223: HH:MM:SS bromen in 2.2.2

Rhumb-Runner, 2023-11-24:


godind, 2023-11-24:
Hi. Thanks for the report.

I not sure I follow completely. May I ask what widget, path, value  type and if you are using unit conversions.

This help us reproduce the problem troubleshoot. We recently fixed an issue that was very similar.

godind, 2023-11-24:
Also. Have you tried the new date widget. It might be better suited to you needs?

godind, 2023-11-26:
Hi,

I've tried but I cannot reproduce this issue. I'll need more details. See above.

Thank you

godind, 2023-11-26:
Actually, I'm 90% certain the Date Value Display Widget will fix your problem. Let me know and I'll close.

Thanks

RhumbRunner, 2023-12-02:
[image]
[image]

Date Value widget doesn't work with the values my timer module (node-red) puts out. The path is not even available to the widget. The path just has the numerical value of the seconds left.  I like the new timer widget but I have a slightly different module that also does pursuit starts. When I go into KIP's data browser and change the units to HH:MM:SS it's correct.

[image]
[image]

godind, 2023-12-02:
As a side note, the date time widget will only work with valid SK date format.

Maybe there is bug in the time transform to hhmmss. Make sure the value is a valid number in seconds with no decimals.

godind, 2023-12-02:
Ok. I got it now. I'll check why the value on 600 is not working with hh:mm:ss. This will take a some time.

PR #241 (2023-12-02): HH:MM:SS broken in 2.2.2
Incompatible number decoration function when string is provided. Fixes #223
commit 6500279f (2023-12-02): HH:MM:SS broken in 2.2.2
Fixes #223
closed 2023-12-02 by godind