mxtommy/Kip #224: Broken font links in Safari

JonyIvy, 2023-11-24:
This is minor and I only see it in Safari, but I noticed some error message in the browser console, four fonts could not be loaded:
```
/@mxtommy/webfonts/fa-solid-900.woff2 (404)
/@mxtommy/webfonts/fa-solid-900.ttf (404)
/@mxtommy/webfonts/fa-regular-400.woff2 (404)
/@mxtommy/webfonts/fa-regular-400.ttf (404)
```
(indeed, these files don't exist)

In the Network tab I can see that the fonts were actually loaded successfully from another path
```
/@mxtommy/kip/assets/fontawesome6/webfonts/fa-solid-900.woff2
/@mxtommy/kip/assets/fontawesome6/webfonts/fa-regular-400.woff2
```

That's why everything is shown correctly, but there must be broken links somewhere in the app.

godind, 2023-11-26:
Hi,

I tried with Safari Version 17.1 (19616.2.9.11.7) but I have no error in the console. Neither with Chrome or Chromium Version 89.0.4389.114.

godind, 2023-12-02:
I upgraded to the latest MacOs version and still can't reproduce.  Can you clear your browser cache and see if it fixes the problem?

JonyIvy, 2023-12-05:
Cleared cache and installed latest KIP. In Safari still the 4 font links return 404. In Firefox these two:
/@mxtommy/webfonts/fa-solid-900.ttf
/@mxtommy/webfonts/fa-solid-900.woff2
Also tried incognito mode, same thing.
Very strange but no real impact.

godind, 2023-12-06:
I found where it is (index.html). It's generated by Build command and I can't figure out how to adjust it or what pushing this <style> tag. Read the AwesomeFont guides but no success.

godind, 2023-12-09:
I think I've fixed this but since I can't reproduce (I know it wierd!), can check with v2.4 please?

Thanks

closed 2023-12-21 by godind

JonyIvy, 2023-12-21:
Still happening.