mxtommy/Kip #234: an enhancement

petsicola, 2023-11-26:
- Add graphical next waypoint mark on wind gauge

petsicola, 2023-11-26:
- Add graphical current on wind gauge

godind, 2023-11-26:
Editing your issue as it's a partial duplicate of #82. Will rename and keep it open as it's a good idea! I suggest you comment on #82 if you want to be notified when released.

Thank you for the feedback! I hope you enjoy KIP!

petsicola, 2023-11-27:
Perfect!

ignamendi21, 2023-12-20:
Is this "water current Flag" implemented????😮

godind, 2023-12-20:
Hi,

We've not started working on this yet. It's on our list. When it gets done this issue will be linked, closed and you will be notified.

KIP is built by volunteers on spare time. It will come, at some point!

godind, 2023-12-20:
Awesome. Let me know when it's ready and I'll see my schedule to include the UI in the widget.

As a side note: check that you send null if your derived calc can't generate path data values for whatever reason, like missing path value dependencies. It's overlooked in many cases and I've seen empty strings, undefined or no output at all witch makes trouble shooting hard from KIP.

Have fun!

davidsanner, 2024-12-20:
If I may add in on this wind gauge thread, I was noticing how much space there is in the middle of the gauge and thinking how a numerical value w/ a large font (user defined path, eg. wind speed, vmg, etc) would be useful and readable from further away like the wind overall direction is readable at a distance (unlike wind speed values which are harder to see at distance).
Another option might be data fields (numerical values) in the corners?

Just another thing to put on the list ;) ... as the app just keeps getting better!