mxtommy/Kip #235: gitignore includes /public

rezder, 2023-11-26:
Public was added to gitignore on 2953ec6 two weeks ago. Is that on purpose.

godind, 2023-11-26:
Yes. Is it a problem?

rezder, 2023-11-27:
Ok I vote for it to be changed back.

When I merged I was left with a suprising few untracked files. 
According to git:

"pull and git merge will stop without doing anything when local uncommitted changes overlap with files that git pull/git merge may need to update".

I hope that cover my case, anyway I have a copy. No harm done.

When change back I assume my now tracked files will still be tracked and I have to remove the tracking with git rm --cached
as gitignore is only for new files I think.

Good luck

godind, 2023-11-28:
I'm no git expert but I don't know why we would to track those files. They local build outputs. Right?

They get generated on the fly and packaged with npm publish so no need to save to the repo?

godind, 2023-12-01:
s

closed 2023-12-01 by godind

rezder, 2023-12-01:
I would have liked an anwser on my guestion and maybe som guidens on what to do. It is the generated folde of ng should I mark it skip-worktree?