mxtommy/Kip #240: Feature request: Make the old Boolean Widget available and other types...

RhumbRunner, 2023-12-02:
I use the Boolean Widget for buttons that don't have a state that the status needs to be tracked like the new widget has. The old widget was fine for my purposes but it just needed the text sizing to work right. The text didn't size to the button.

Can we ultimately have multiple styles of Boolean widgets. I see three possible uses: 
1) Momentary... Just a visual state change when PUSH is sent. 
2) Latching... The Button visual state (true / false) is tied to the same path as the PUSH?
3) Modal... like the v2 style with the separate visual indicator that could be different colors based on a _different_ path than the PUSH.  ie. a button that steps though several states. The indicator can let you know which state the system is in.

godind, 2023-12-02:
Hi,

Thanks for the feedback. I would love to get this right. For some reason it's become a popular widget in the last month or so. I did an ugly patch for v2 as it was not really usable based on user feedback. The current switch/button widget should support momentarily and state mode in its Options settings. Have you had a look?

Can you share what you use those button for? It would help scope things out.

- Momentary: this we have right?
- Latching: this also?
- Modal: not sure about the use case.

I also think some user would like a single widget that groups a bunch of light - label - button together. Similar to a breaker panel. Easier to arrange and position instead of having, say,  six widgets on a page. But that's a bit more work to implement the dynamic Options settings and rendering.

If see few different use cases.

The feature backlog is increasing! I love it.

RhumbRunner, 2023-12-03:
Examples of how to handle button text can also be used for widget labels...

[image]

godind, 2024-01-18:
see 2.7 release shortly. Thanks

commit 8f633785 (2024-01-17): Feature request: Make the old Boolean Widget available and other types...
Fixes #240
closed 2024-01-18 by godind