mxtommy/Kip #248: Error: Unable to convert "--" into a date (version 2.4.0)

Techstyleuk, 2023-12-10:
I am running 2.4.0 and was making some updates to the ETA calc (eta.js) in Derived Data plugin and first I found that KIP now sees 'navigation.courseGreatCircle.nextPoint.estimatedTimeOfArrival' as a date and that is awesome and works great!!  Eg:
[image]

The issue is that when there is no speed there is no time of arrival and in the calculation I set the value to "--".  This now shows an error
[image]

Can KIP be changed to handle this or what should I set the value to be?

godind, 2023-12-10:
Based on the specs and discussion with the team, path values can either be a specific value or `null`. `null` should be used when no data can be produced/generated, such as when a depth sounder is in ocean level depth - no bottom reading. So I suggest you send `null` when you can't produce values.

In the case of the Date widget, it will only accept a string that parses to rfc3339 date format (SK's standard datetime) and that converts to a Date type. The widget will render `null` values as `--`.

But I'm having doubts about the way I filter data streams in KIP... try and I'll fix it if the `null`s get filtered by mistake by that widget.

Techstyleuk, 2023-12-10:
I just tried it and it shows null in SignalK data browser but it does not change anything in KIP, it ignores the null and maintains the value before. Although, once the timeout kicks in the value goes to --

godind, 2023-12-11:
I'll fix it so null are not filtered. Soon...!

closed 2023-12-12 by godind

godind, 2023-12-12:
As of 2.4.1 it should now receive `null` and display `--`.

Let me know...

PR #251 (2023-12-12): Error: Unable to convert "--" into a date (version 2.4.0)
Fixes #248