mxtommy/Kip #252: save configurations.

petsicola, 2023-12-13:
add the ability to download all configurations and save it in another mxtommy kip. I change my signalk server and i need to backup and restore all the configurations in the new server. I dont know how to make that

godind, 2023-12-13:
If you backup you SK config, or you upgrade the server, the KIP config will stay as-is. 

The SK config is located in the user folder that is running the process. Normally under ~/.signalk

godind, 2023-12-13:
All app config are under `~/.signalk/applicationData`. KIP Global config is under `~/.signalk/applicationData/global/kip` and user config under  `~/.signalk/applicationData/<username>/kip `

godind, 2023-12-19:
Get the latest master branch

- npm i
- Ng build
- ng serve --configuration=dev --serve-path=/@mxtommy/kip/

Point your browser to 
- localhost:4200/@mxtommy/kip/

It should run

petsicola, 2023-12-20:
Perfect. I copy the config to a new device with open plotter and I have my config in the new device.
Thank you again.

[image]

godind, 2023-12-23:
Great! Thanks for sharing the screenshot. It's fun and surprising to see how KIP is being used. It helps defining the approached for the design.

Enjoy and share your feedback!

petsicola, 2023-12-23:
I think that this is one of the most powerfull plugins in signalk. I try to use this config to know the performance of my boat. I like your work. Thank you again.

godind, 2023-12-24:
What plugin is that? It's a performance plugin? I was looking to improve the wind widget to include performance data and polar performance kind on of info.

petsicola, 2023-12-24:
mxtommy kip is one of the best plugins of signalk ;).

godind, 2023-12-24:
Would it be possible for you to record SK data while racing or practicing? It's winter here in Canada. That would help to build and test the widgets.

A few quick upwind tacks, downwind jibes, up/down wind mark rounding with racing environmental data in SK stream file would be very useful.

The stream file I have is a sad replacement for a race simulation.

petsicola, 2023-12-24:
no problem, i´m going to try to save the data in video this week. 
I don´t know if i have another video.

https://github.com/mxtommy/Kip/assets/61534583/1e0d7dd2-4e05-4c6b-9504-3bf31cf725b5

petsicola, 2023-12-24:
I had a VDR and I captured this video

https://github.com/mxtommy/Kip/assets/61534583/4f8f3e75-23fd-4df1-817e-802188c5ac46

godind, 2023-12-24:
Oh. I forgot one parameter to enable in SK for data logging: you also need accessLogging enabled in Server / Settings. I've updated the instructions...

godind, 2023-12-24:
We should probably move conversation this to a new Issue or to Slack ;)

closed 2023-12-26 by godind

petsicola, 2024-01-01:
Hi.
I made the data logging and I have a file with this data. But when I try a new filestream conecction in signalk server I have an error. 

**Filestream TYPE Multiplexed LOG**
1703948400017;A;2023-12-30-15:00:00.017,2,127245,172,255,8,fc,f8,ff,7f,ff,7f,ff,ff
1703948400021;A;2023-12-30-15:00:00.020,4,129038,4,255,8,00,1b,c1,56,39,77,0d,4f
1703948400021;A;2023-12-30-15:00:00.021,4,129038,4,255,8,01,39,b9,fa,68,05,2c,19
1703948400022;A;2023-12-30-15:00:00.021,4,129038,4,255,8,02,eb,df,b1,92,02,00,00
1703948400023;A;2023-12-30-15:00:00.023,4,129038,4,255,8,03,00,ff,ff,ff,7f,f0,fe
1703948400037;A;2023-12-30-15:00:00.037,2,127250,204,255,8,ff,07,42,ff,7f,ff,7f,fd
1703948400037;A;2023-12-30-15:00:00.037,2,127251,204,255,8,ff,5c,18,a1,ff,ff,ff,ff
1703948400037;A;2023-12-30-15:00:00.037,2,127257,204,255,8,ff,07,42,7d,01,cb,fb,ff
1703948400038;A;2023-12-30-15:00:00.037,2,127245,204,255,8,fc,f8,ff,7f,ff,7f,ff,ff
1703948400038;A;2023-12-30-15:00:00.038,2,127245,204,255,8,00,ff,ff,7f,ff,7f,ff,ff
1703948400086;A;2023-12-30-15:00:00.042,2,130306,115,255,8,00,7b,03,90,36,f2,ff,ff
1703948400086;A;2023-12-30-15:00:00.049,2,129025,3,255,8,35,8c,2b,19,46,a0,c4,fa

**Signalk Error log** 
Dec 30 15:28:31 2023-12-30-14:00:39.769,2,127251,204,255,8,ff,c6,62,15,00,ff,ff,ff Parser not found for input. - 2023-12-30-14:00:39.769,2,127251,204,255,8,ff,c6,62,15,00,ff,ff,ff
Dec 30 15:28:31 2023-12-30-14:00:39.770,2,127257,204,255,8,ff,f5,dc,69,01,fd,03,ff Parser not found for input. - 2023-12-30-14:00:39.770,2,127257,204,255,8,ff,f5,dc,69,01,fd,03,ff
Dec 30 15:28:31 2023-12-30-14:00:39.770,2,127245,204,255,8,fc,f8,ff,7f,ff,7f,ff,ff Parser not found for input. - 2023-12-30-14:00:39.770,2,127245,204,255,8,fc,f8,ff,7f,ff,7f,ff,ff
Dec 30 15:28:31 2023-12-30-14:00:39.771,2,127245,204,255,8,00,ff,ff,7f,ff,7f,ff,ff Parser not found for input. - 2023-12-30-14:00:39.771,2,127245,204,255,8,00,ff,ff,7f,ff,7f,ff,ff
Dec 30 15:28:31 2023-12-30-14:00:39.771,7,65359,204,255,8,3b,9f,ff,ff,ff,65,d2,ff Parser not found for input. - 2023-12-30-14:00:39.771,7,65359,204,255,8,3b,9f,ff,ff,ff,65,d2,ff

godind, 2024-01-01:
Feels like an SK issue. Ask in the SK support forum. You're on Slack, right?

Try a few different stream settings maybe?

petsicola, 2024-01-02:
no, I´m not in slack, Wich is the slack channel?
Could you invite me to the channel?

godind, 2024-01-03:
I'll. Give your email.

It's the Signalk-dev Slack workspace. It has a bunch of Chanel on different subjects. One of them is KIP.

Response is pretty good. Usually someone helps within 2-4h. Worst case is 24-48h

godind, 2024-01-04:
I meant your email address so I can send you a Slack invitation. Write it here or:

[try to click this link](https://join.slack.com/t/signalk-dev/shared_invite/zt-1leccop43-KrU7G6yBq9g91KXjZtNg1g)