mxtommy/Kip #254: Godind/issue176 #228  removed a null check

rezder, 2023-12-18:
A null check was removed from line 278 in src/app/signalk-delta.service.ts
changing from 
if ( (typeof(item.value) == 'object') && (item.value !== null)) {

to             if (typeof(item.value) == 'object') {

null is is object  and the use of Object.keys() produces a lot of errors in my consoll.

What should prevent this value from being null now.

godind, 2023-12-18:
Awesome! Thanks for investigating this issue. Did you run some tests or does this require deeper validation?

rezder, 2023-12-19:
Sorry I be more precis next time. I got the nulls from my GPS when using it indoors. The code looks like it have been handling Nulls before and there is null handling in widgets. I thought something was overlooked when it was removed and I was hoping you would fix it easily as you just changed it resently. It is not a solution you need find out if the null should be there or not, I did not look in to it.

godind, 2023-12-19:
No worries. It got me set on the right track to find the problem quickly as I had just received a data stream file from another use on the same problem.

It led me to other cascading issue all the way up to the widget option source selection. So thanks for the insight!

Are you able to build from master and validate that it's fixed and that you don't see another problem be I release a version?

I've done some testing but we never know!

rezder, 2023-12-19:
Just got something up and running today I can build on my arch linux  dev computer and test it on my boat computer raspberry pi with my old gps. The ng http server part is still not working but it build without problems. What should I pull down and be precis as I am not messing around on github everyday.

godind, 2023-12-20:
Another user tested it reported the issue fixed. I'll release 2.4.2 shortly.

godind, 2023-12-20:
If ever you want to build and run KIP you can look at the doc but the short version is:
- get master branch
- npm install
- ng build
- npm run dev

Point your browser to http://localhost:4200/@mxtommy/kip

closed 2023-12-20 by godind

rezder, 2023-12-20:
I have gone throught the last 2-3 weeks of null changes I think there are some mistakes.

On 26/11 #229 on a typo issue:  
in src/app/widgets/svg-wind/svg-wind.component.ts 
You added alot of these null returns
```
if (changes.compassHeading.currentValue === null) {return}
```
Why? and I do not think they are need any longer with the data stream setting every thing to zero on #242

I ask because that is "ignore error in data code" and like to know what error you prevent here and not earlier.

On 5/12 #242 Time out was introduced 
If I understand the code correctly the function timeoutPathObservable  returns the value 0 and state 0 for numbers but a the code and in comment i see this:

```
this.observeDataStream('appWindSpeed', newValue => {
      if (newValue.value == null) { // act upon data timeout of null
        newValue.value = 0
      }
      this.appWindSpeed = newValue.value;
    });
```
looks like it believe the value returned is null.

My questition is, does the data stream return null or zero for numbers when timeout happens.

On 8/12 #244 waypoint was introduced with no null check 
in /widget-wind.component.ts
```
this.observeDataStream('nextWaypointBearing', newValue => {
      if (newValue.value < 0) {// stb
        this.waypointAngle = 360 + newValue.value; // adding a negative number subtracts it...
      } else {
        this.waypointAngle = newValue.value;
      }
    }
  );
```
This will send null value on but why not handle it the same way as the others.

on 12/12 #251 the hounds was released data streams now have nulls makinge 
Null to zero code valid with out the comment and need on waypoint.

godind, 2023-12-23:
The approach changed a few times as we tackled a combination of different issues coming from different requirements and new features such as support for: sample time, data expiration, source priority in combination with additional widget features and use cases.

The final design regarding `null` is:
-  SK supports `null` values so KIP should accept them. When path sources can't generate values; like a functioning depth sensor that goes into very deep water, it will stop generating values because the sensor can't hit bottom. As per SK specs, `null` is not an error. It means `No value can be generated`. 
- KIP TTL feature: If data expires, KIP will replace stale data values in the internal data tree with `null`, until SK starts to send new data again.
- Each individual Widget needs to handle `null` according to their unique use cases.

The challenge is that not all widget features, or even all the individual paths within a widget, can handle `null` values the same way. It comes down to the use case. Some can simply display `--`, while others do calculations that impact some others variables used elsewhere. So in some case it's simpler just to stop processing, or hide a UI component, etc. It depends... Say for Waypoint in svg, if it's null, we simply don't display the UI component.

That being said, it's not impossible some bugs are present or that improvements are required. KIP would greatly benefit from having test automation. But no one wants to work on this ;)

I am very glad for your interest in improving KIP. Thank you for your involvement. That said, I think we can improve on the collaboration: making replies to old closed PRs and closed Issues is hard to follow, to decipher and to revers engineer. The approach probably generates disappointment on your side and it fells like a critique of past work on our side.

Here is what I propose:
- Reach out on Slack "Signal-dev / #kip channel, or directly to me or Tommy" to talk over things and share ideas. With over 1000 KIP users, everyone has its own opinion, requirements and personal views on what and how KIP should do things. I find that discussing goes a long way to level perspectives and make for a strong product.
- Before starting work on KIP, validate the approach, intent or design ideas if you don't want to waist precious personal time.
- Always start the discussion from the latest master branch version or an uncommitted PR.
- If you are fixing a bug, adding a feature, improving code, etc. first create a KIP Issue.
- Contribute and propose code changes using PRs against the master branch. Link the PR to your Issue. PRs will be reviewed and merged when we get to it. Reviews can require you to adapt and fix things, or they can be refused. If we have spoken prior, it should be smooth. We've had successful contributions in the past. They are very much welcomed!

So, for your current reported comment on `null` handling in Wind widget, or themes problems, I invite you to submit PRs.

Hope this helps.
Enjoy the Holidays

rezder, 2023-12-25:
Appreciate you reaching out it is solid thing to do. I have been rethinking my approach when someone imply that you do not know the difference between a ring and a disable button you must be able to do better.

The problem I was having was the I rebased my branch and to do that I looked at the changes that was made that effected me and did a natural code review and found some issues. Broken assumption failed logic but I do not know the the frame work angular and rx or even js. I can not make issue or a PR before I trace and test to learn the frameworks. So I have just dump those issue where I though the may be use full or some one could answer the question they poses. I did not expect any one to care. If an issue is important enough to me I should trace it down and make an issue.
 
Like the null think I noticed the the code sat the value to zero for numbers and a lot null exception I wanted to make that comment somewhere and know why. I did not want to make the PR because I do not know if rx and angular adds nulls. The code is changed again and I may have failed to see that in time. I will keep the latest master and stop making code comment. My issues are fading anyway and soon I expect only the one I care about remains. I was planing to make them in to issues but it is effort I do not care for.

As for making PG to change the null handling remove the unnecessary null exception and remove comments that make no sense to me. I can not imagine that will offend less than a few comment on old commits.  A Quick look in widget-wind.component.ts tell me all datastream gets there null value set zero except waypoint so any later fine handling of null in svg-wind as you describe might be challenging. The setting of zero is followed by the comment "act upon data timeout of nulI" which I hate because it imply null comes only with time out. I fear the angular post change frame work and  removing other people comment No thank you, I stay away from that one.

As for the missing numbers on the autopilot I could fix that but You are talking about Material styles and angular 16+ I do not know what that is but removing the hard coded colors adding an extra class for the numbers I can can do that. I would be a god simple PR to learn on I do not have more of issue like that.

I assume I just update my fork  branch it and  make the change and then create a pull request. I will not do any thing before you confirm.

After that I rebase my branch if you could stand still for a second :) make some new widget and decide on what to do:
Stay away from KIP or try to get some changes through, I now know
a little more on how to do that thank you for that.

I consider joining slack but it is not a good fit with my system.

Sorry for the inconvenience I caused. Have a nice holiday.

Ps You do not need to defend the development of the code I am old enough to know how it is made
and not old enough to not make errors my self. I will never make a comment to point a finger.

godind, 2023-12-26:
To contribute to KIP you can read the projet readme file for instructions. I am not sure about the state of your fork but I recommend deleting it and starting over from scratch to eliminate any issues. Delete your repository and fork again.

When you work, you always work from your fork. You should also not work on your fork's master branch.

To contribute, create a new feature branches from your fork's master.

Once you are done with your feature branch, commit your changes and publish to your fork.

You can then create a PR and add some details. The PR will come to
KIP.

Once it gets merged onto KIP, you can delete both your local and remote feature branch.

Then update your forks master branch, you will see the PR changes as part of the master update.

Hope this helps.

Happy Holiday